mavlink/qgc-user-guide #29: develop an Android APP like QGC to connect to PX4 JMAVSim simulator and take off

seantobe, 2016-09-13:
Dear all:
I am trying to develop an Android APP that can connect to PX4 JMAVSim simulator and take off. As I turn on the JMAVSim, and then turn on QGC, they are automatically connected. If anybody knows how, please reply me, thanks a lot.

BR
Sean Hsu

DonLakeFlyer, 2016-09-13:
It connects over udp: https://github.com/mavlink/qgroundcontrol/blob/master/src/comm/LinkManager.cc#L474

closed 2016-09-13 by DonLakeFlyer

seantobe, 2016-09-14:
Hi DonLakeFlyer:
I found the sample Code for Ardupilot software, and made some modification and can successfully compiled and run the APP. However, when I press the Connect button to connect through udp, it does not seem to connect. First of all, I am not sure whether this sample APP can connect to PX4 software or not. If you know please let me know, I really appreciate it. I posted my APP code as below.

===============================================
package com.o3dr.hellodrone;

import android.os.Handler;
import android.support.v7.app.ActionBarActivity;
import android.support.v7.app.AppCompatActivity;
import android.os.Bundle;
import android.view.View;
import android.widget.AdapterView;
import android.widget.ArrayAdapter;
import android.widget.Button;
import android.widget.Spinner;
import android.widget.TextView;
import android.widget.Toast;

import com.o3dr.android.client.ControlTower;
import com.o3dr.android.client.Drone;
import com.o3dr.android.client.apis.ControlApi;
import com.o3dr.android.client.apis.VehicleApi;
import com.o3dr.android.client.interfaces.DroneListener;
import com.o3dr.android.client.interfaces.TowerListener;
import com.o3dr.services.android.lib.coordinate.LatLong;
import com.o3dr.services.android.lib.coordinate.LatLongAlt;
import com.o3dr.services.android.lib.drone.attribute.AttributeEvent;
import com.o3dr.services.android.lib.drone.attribute.AttributeType;
import com.o3dr.services.android.lib.drone.connection.ConnectionParameter;
import com.o3dr.services.android.lib.drone.connection.ConnectionResult;
import com.o3dr.services.android.lib.drone.connection.ConnectionType;
import com.o3dr.services.android.lib.drone.property.Altitude;
import com.o3dr.services.android.lib.drone.property.Gps;
import com.o3dr.services.android.lib.drone.property.Home;
import com.o3dr.services.android.lib.drone.property.Speed;
import com.o3dr.services.android.lib.drone.property.State;
import com.o3dr.services.android.lib.drone.property.Type;
import com.o3dr.services.android.lib.drone.property.VehicleMode;

import java.util.List;

public class MainActivity extends ActionBarActivity implements DroneListener, TowerListener {

    private Drone drone;
    private int droneType = Type.TYPE_UNKNOWN;
    private final Handler handler = new Handler();
    Spinner modeSelector;
    private ControlTower controlTower;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_main);

        // Initialize the service manager
        this.controlTower = new ControlTower(getApplicationContext());
        this.drone = new Drone(getApplicationContext());

        this.modeSelector = (Spinner)findViewById(R.id.modeSelect);
        this.modeSelector.setOnItemSelectedListener(new Spinner.OnItemSelectedListener() {
            @Override
            public void onItemSelected(AdapterView<?> parent, View view, int position, long id) {
                onFlightModeSelected(view);
            }
            @Override
            public void onNothingSelected(AdapterView<?> parent) {
                // Do nothing
            }
        });

[... 218 lines cut ...]
    }

    public void onArmButtonTap(View view) {
        Button thisButton = (Button)view;
        State vehicleState = this.drone.getAttribute(AttributeType.STATE);

        if (vehicleState.isFlying()) {
            // Land
            //this.drone.changeVehicleMode(VehicleMode.COPTER_LAND);//changeVehicleMode is out of date, use the following function instead
            VehicleApi.getApi(this.drone).setVehicleMode(VehicleMode.COPTER_LAND);
        } else if (vehicleState.isArmed()) {
            // Take off
            //this.drone.doGuidedTakeoff(10); // Default take off altitude is 10m
            ControlApi.getApi(this.drone).takeoff(10, null);
        } else if (!vehicleState.isConnected()) {
            // Connect
            alertUser("Connect to a drone first");
        } else if (vehicleState.isConnected() && !vehicleState.isArmed()){
            // Connected but not Armed
            //this.drone.arm(true);//arm is out of date, use the following function instead
            VehicleApi.getApi(this.drone).arm(true);
        }
        Toast.makeText(getApplicationContext(), "onArmButtonTap", Toast.LENGTH_LONG).show();
    }
}