mavlink/qgc-user-guide #65: Doc Bug: Daily Build New Features - Needs explanation

hamishwillee, 2017-05-04:
This is named "*Daily Build* New Features" but it isn't clear what it contains. 

@DonLakeFlyer Is this a log of new features that have gone into the daily build since the last official release ([Release Notes 3.1](https://docs.qgroundcontrol.com/en/ReleaseNotes.html))?

Bug Page: [Daily Build New Features · QGroundControl User Guide](https://docs.qgroundcontrol.com/en/DailyBuildChanges/DailyBuildNewFeatures.html)

DonLakeFlyer, 2017-05-05:
No there is no log. I had been trying to just add headers to that daily build page but I'm sure I missed a ton of stuff. I'll need to do a full review when we get ready for next Stable which is coming up fast.

hamishwillee, 2017-05-10:
This will be improved by #70 , which adds the following line to the top:

  This topic contains a high level and *non-exhaustive* list of new features added to *QGroundControl* since the last [stable release](../releases/release_notes.md). These features are available in [daily builds](../releases/daily_builds.md).

hamishwillee, 2017-10-23:
Addressed

closed 2017-10-23 by hamishwillee