mavlink/qgc-user-guide #72: Add information about Sub

hamishwillee, 2017-05-10:
We minimally need "airframe setup" information. Not sure how to handle this - depends on how firmware specific the support is, and how different it is from other vehicles - e.g. can we plan missions, can we do other things ...

hamishwillee, 2017-05-10:
FYI some support added in https://github.com/mavlink/qgroundcontrol/pull/5121

@jaxxzer @DonLakeFlyer if there is specific information you'd like added about sub support in QGroundControl user guide, please let me know. I need an overview of what support we have before I can work out a good way to integrate.

jaxxzer, 2017-05-25:
Hi @hamishwillee the vehicle setup stuff is the only thing that's really important/different for general sub users at the moment, and most of what's there that's sub specific, we have documented elsewhere. If you would like to document it here, I am happy to act as a reference.

Anything GPS-related with sub is still in development. The Plan view does work with Sub, but most will not be able to use it due to cost-prohibitive hardware restrictions.

hamishwillee, 2017-06-09:
Thanks. Let's do some docs when the "frame" appears in a daily build/stable release.