mavlink/qgc-user-guide #111: Camera drop-down additions: Ricoh GR II and MicaSense RedEdge

Otusbubo, 2017-10-10:
Ricoh GR II
fl (mm): 18.3
image width (pix): 4928
image height: 3264
sensor width (mm): 23.7
sensor height: 15.7

RedEdge
fl: 5.5
image width: 1280
image height: 960
sensor width: 4.8
sensor height: 3.6

Triple checked!  Plus, the ability to "save as" to create/save our own cameras would be fantastic.  Thanks!

hamishwillee, 2017-10-10:
@Otusbubo Is this existing functionality that you think should be documented or new functionality that you want added to QGC? If it is the later (as I expect) can you please repost in https://github.com/mavlink/qgroundcontrol - this repo is for user guide additions.

DonLakeFlyer, 2017-10-11:
I had asked @Otusbubo to put in an issue with the camera specs so I can add them to QGC. He put the issue in the wrong place.

hamishwillee, 2017-10-11:
Replaced by https://github.com/mavlink/qgroundcontrol/issues/5740

closed 2017-10-11 by hamishwillee