mavlink/qgc-user-guide #136: Add very cool PID tuning stuff

hamishwillee, 2018-04-03:
Add docs: https://github.com/mavlink/qgroundcontrol/pull/6295

jaxxzer, 2019-01-22:
There is also a tuning page available for ArduSub. We have yet to add support for ArduPilot generally, but I think it would be straightforward.

Note the ardusub tuning page is not quite as sophisticated as this support for PX4, but we could work on that too.

hamishwillee, 2019-01-31:
@jaxxzer 

> We have yet to add support for ArduPilot generally, but I think it would be straightforward.

I presume you mean this in response to Don's comment below

> The above currently is only supported for PX4 Pro firmware and multi-rotors. But that said the whole thing is implemented in a generic fashion such that it just works off of generic lists of parameters. So all I need to do is figure out what all the other param sets are to implement other vehicle types and ArduPilot support.

Should the sub page be implemented using this new feature instead? How do we get the params Don needs for this for the various ArduPilot vehicle types?

Can you recommend best person to document the tuning page for Sub?I don't know enough about Sub tuning to do a good job but I'd happily subedit.

closed 2019-02-03 by hamishwillee

jaxxzer, 2019-02-12:
Hi there @hamishwillee. @Williangalvani and I will be able to help you with anything sub-related.

hamishwillee, 2019-02-12:
Thanks @jaxxzer !

PR #175 (2019-02-03): [WIP] Add PX4 advanced tuning info.
Fixes #136 
(ie adds support tor PX4 advanced tuning https://github.com/mavlink/qgroundcontrol/pull/6295)

Note, I may split into separate ArduPilot and PX4 docs, which is why this is still WIP.