mavlink/qgc-user-guide #138: Clarification of battery failsafe needed

JonReacher, 2018-04-28:
Critical and emergency battery thresholds parameters state "This threshold commonly will trigger RTL." It would be clearer if it states that it will follow the failsafe selected in the Safety setup. Or am I wrong and there is some override of the desired failsafe, and it goes to RTL in certain instances?

DonLakeFlyer, 2018-04-29:
This text comes from the firmware. So any change to it or review needs to be done by firmware folks. Needs to be posted in firmware issues.

closed 2018-04-29 by DonLakeFlyer

hamishwillee, 2018-04-30:
@JonReacher You are both right and wrong :-). The safety action is a [selector](https://docs.qgroundcontrol.com/en/SetupView/Safety.html), so you can choose whether you want to warn or RTL or whatever. Where you are right is that the text is confusing, since the **default** action default is warning. Further discussion in https://github.com/PX4/px4_user_guide/issues/205#issuecomment-385292448