mavlink/qgc-user-guide #143: document the mavlink console

bkueng, 2018-06-21:
I often need to point people to the mavlink console, so it would be good to have instructions and a screenshot showing how to access it.

I see that all Items on https://docs.qgroundcontrol.com/en/toolbar/toolbar.html have a separate page already except for the Analyze tab.

hamishwillee, 2018-06-22:
Yes. There is actually a broader requirement to go through and integrate all the changes that were added in [3.2](https://docs.qgroundcontrol.com/en/releases/stable_v3.2_long.html). Don needs help with that, but has not previously been a priority. I hope to find time in coming months, management permitting.

DonLakeFlyer, 2018-06-22:
And now 3.3 as well which will be going out the door soon.

hamishwillee, 2018-07-17:
Fixed by https://github.com/mavlink/qgc-user-guide/pull/145

closed 2018-07-18 by hamishwillee