mavlink/qgc-user-guide #184: Ubuntu dependencies: documentation mentions github readme, but github readme does not contain these dependencies

thomasgubler, 2019-02-13:
Under https://docs.qgroundcontrol.com/en/getting_started/download_and_install.html under Ubuntu Linux / Compressed Archive it is mentioned to "Install additional packages as specified in the github README. You do not need to install Qt."

However the github README has been stripped of mentioning these dependencies.

hamishwillee, 2019-02-14:
@DonLakeFlyer Do we still supply the compressed archive? If so, can you let me know what additional packages are required to use it on Ubuntu?

DonLakeFlyer, 2019-02-15:
> Do we still supply the compressed archive?

Not really. Only AppImage is fully supported.

hamishwillee, 2019-02-18:
>> Do we still supply the compressed archive?

> Not really. Only AppImage is fully supported.

@DonLakeFlyer So do I remove it then? I saw a post recently where someone was using it with GStreamer, so I am worried perhaps it is "needed".

If we are not deleting we need to supply the dependencies - can you do that?

DonLakeFlyer, 2019-02-18:
> So do I remove it then?

Remove it.

hamishwillee, 2019-02-19:
Thanks. Done #189

PR #189 (2019-02-19): Remove appimage from downloads
Fixes #184 by removing appimage option - as suggested by Don here: https://github.com/mavlink/qgc-user-guide/issues/184#issuecomment-464922985
closed 2019-02-19 by hamishwillee