mavlink/qgc-user-guide #209: QGC v3.5.0 Won't Open in Windows 10 Pro v1803

rollysing, 2019-03-31:
Just installed QGC v3.5.0 in Lenovo laptop running Windows 10 Pro v1803. QGC attempts to open but fails. While looking at Task Manager, the app starts to open but Status momentarily shows Suspended, then QGC disappears.
- Have disabled signature requirement during startup prior to installation.
- Have tried running Safe Mode and Compatibility versions but still no joy.

Any ideas/ advise on how to resolve is highly appreciated. @DonLakeFlyer any thoughts or comments?

Thanks.

rollysing, 2019-03-31:
By the way, as of today, I also tried installing the latest Daily Build and am got the same result as described above.

Thanks.

rollysing, 2019-03-31:
Here is a short video of what happens in the Process tab of Task Manager:
https://youtu.be/HfB_WZuV08Q

Thanks.

DonLakeFlyer, 2019-04-19:
My guess would be that it might be crashing at startup. 

"When QGC crashes a crash dump file will be place in the Users LocalAppData directory. To navigate to that directory use the Start/Run command. You can bring this up window WinKey+R. Type into that %localappdata% for Open and click Ok. Crash dumps will be in a QGCCrashDumps folder in that directory. You should find a new .dmp file there. Add a link to that file in a GitHub Issue when reporting you problem."

DonLakeFlyer, 2019-04-19:
Also @rollysing the correct place for QGC Issue is here: https://github.com/mavlink/qgroundcontrol/issues. We can continue to track this specific on here fo rnow.

rollysing, 2019-04-19:
Opened and to be continued at the recommended location:
https://github.com/mavlink/qgroundcontrol/issues/7367

closed 2019-04-19 by rollysing