mavlink/qgc-user-guide #210: Stale footer on main page

jaxxzer, 2019-04-19:
The 'news' and the copyright are stale:
[image]
[image]

hamishwillee, 2019-05-01:
Thanks @jaxxzer 

@mrpollo @DonLakeFlyer I have updated the footer copyright to 2019 - I'd have done a range, but not sure of the right ones for Dronecode and QGC.

However not sure how you want to deal with the news section? Remove it altogether somehow or do you want to add blog announcements here for new releases so this is up to date?

DonLakeFlyer, 2019-05-01:
Remove seems fine to me.

hamishwillee, 2019-05-01:
OK. Removed. I think this is now done.

closed 2019-05-01 by hamishwillee

mrpollo, 2019-05-03:
Thanks @hamishwillee @DonLakeFlyer