mavlink/qgc-user-guide #275: Developer build issues.

madflower, 2020-08-24:
In the documentation on this page
https://dev.qgroundcontrol.com/master/en/getting_started/
For the qt-pro stuff it says:
In the installer Select Components dialog choose: 5.12.6.

5.12.6 isn't included in the qt-pro package by default. 

I am receiving build errors of: 
Project ERROR: Unknown module(s) in QT: location positioning quickcontrols2 texttospeech multimedia location-private positioning-private charts x11extras waylandclient

hamishwillee, 2020-08-25:
@DonLakeFlyer Leaving these technical queries to you.

DonLakeFlyer, 2020-08-25:
No idea what "qt-pro" is. But the docs also say "If the version needed is not displayed, check the archive (show archive and refresh)."