mavlink/qgc-user-guide #279: Explanation of QGC Console Messages and Error Meanings...?

jimlp12, 2020-10-23:
QGC and Pixhawk 4 Flight Controller.... Quad copter will not arm for flight, messages says, "SEE CONSOLE".....? QGC CONSOLE lists "D", "I", and "E" lines with seems to be errors encountered by either QGC or Pixhawk Flight Controller.....! NO WHERE in any Documentation is there any type of Explanation of these Error's and their Causes.... You can not assume that everyone who uses QGC/Pixhawk knows everything there is to know about either device. I've submitted the problems to PX4.io a supposedly support Discussion and they never replied because they wanted a flight LOG, well people, if your Device/Quad/Plane/Rover/whatever will not ARM.... it's not going ANY WHERE, and therefore A FLIGHT LOG IS NOT CREATED......! So all the Braininess need to come together and focus on the 844 issues - and stop sitting on their thumbs doing nothing....!!!!!!!!
If ya can't fix the problems....get the hell out of the Business...>!      =)

hamishwillee, 2020-10-25:
It sounds like your problem is not arming? That is a flight stack specific question. By default, as you say, no log is generated until you've armed. YOu can change this with the setting shown here: https://docs.qgroundcontrol.com/master/en/SettingsView/General.html#telemetry_logs

[image]

The console log https://docs.qgroundcontrol.com/master/en/SettingsView/console_logging.html is for errors in QGC itself. Typically you use that for unexpected app behaviour - e.g. connection never starts, video not working etc. It can give hints to the inexperienced (me) but generally you just pass that on to the QGC team.

hamishwillee, 2020-10-25:
I'm closing this. I hope your log helps the PX4 team debug.

Just FYI, you can probably also get more debug yourself using the MAVSDK: https://mavsdk.mavlink.io/develop/en/guide/taking_off_landing.html#health-check

closed 2020-10-25 by hamishwillee

jimlp12, 2020-11-02:
Hamish Willee;
I just read the section on MAVSDK, like you I'm new to this Drone field and very inexperienced with no programming background of any kind let alone knowing how to compile some stuff to create a MAVSDK library..? Like Linux, I'm pissed that the lack of documentation for these Open Source Softwares (linux, Pixhawk 4, etc.) where everyone you meet in person or otherwise assumes you know everything about the subject, the issue, the problem, the solution, etc....! I've got a lot of money invested in this Quad and the efing thing doesn't work, I submit an issue and get all kinds of resolves from the know it all people but none of that ever helps, it's like everyone is just guessing....I am ready to trash the whole efing thing and buy a store bought one, that'll at least work out of the box...!
James....

Sent with [ProtonMail](https://protonmail.com) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

jimlp12, 2020-11-02:
Hamish; The console log https://docs.qgroundcontrol.com/master/en/SettingsView/console_logging.html is for errors in QGC itself. (you wrote this part)...to me...! This is what I received on QGC when it said "SEE Console"...! People at Discuss demanded a flight LOG which I said "One (Flight LOG)is not generated unless the Vehicle is ARMED"...! Without a Flight LOG, they couldn't help me...! This is what I mean when trying to troubleshoot and get help from the so-called community..! One guy told me the Console messages are crappy messages from the Vehicle...! AND, that QGC only shows MESSAGES in the MEGAPHONE/YIELD SIGN ICON at the top..! He said he'd try to fix it, I think he was DonLakeFlyer, @Discussion Forum-Pixhawk PX4, QGC, MAVSDK..! I paid $227.94 for my Holybro Pixhawk 4 FC...! and it's f-cked up...? Contacted them, same crap, I need to unlock the PX 4...? first they wrote unlock QGC...? They don't work weekends....! I went to Github, they show QGC with 844 Issues and Mission Planner with 914 Issues...? What the heck is going on.....? If everything is so screwed up, how the heck are we as users, flyers, pilots... supposed to troubleshoot our problems....????
James...

Sent with [ProtonMail](https://protonmail.com) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

hamishwillee, 2020-11-02:
@jimlp12 As you say, if you want to get something that works out of the box then you need to buy a vehicle where everything is integrated. Then there is just one place you ask for help. If you want to use open source then you have to do some integration and digging yourself. 

You have done that, and now you have an answer. The setting I showed you tells you how to turn on logging before arming. http://docs.px4.io/master/en/getting_started/flight_reporting.html shows how to upload a log. With that you can go to the discuss server, and post a link to your log and they can help you debug. 

The QGC people were trying to help you by pointing out that the flight stack is what decides if it can take off or not, and what sends the status messages. Don can't help you with that. All QGC does is display that information.

Good luck!

jimlp12, 2020-11-02:
Hamish; Okay, I will try what you said about turning on Logging before Arming.....?
James.....

Sent with [ProtonMail](https://protonmail.com) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

jimlp12, 2020-11-03:
Hamish;
Good news....! Today, I removed QGC from my Laptop and I later installed Mission Planner GCS(ground Control Station). I then downloaded latest Firmware, which went well and I then Proceeded to go through all the Calibrations of Pixhawk 4, RadioMaster TX16s Transmitter and bound it with Radio Receiver. I did not receive ANY Error's...! I was able to ARM the Quad Copter but I only have 2 Motors out of four that are working. With some jumper wires, Motors 2 & 3 Worked, 3=CW, 2=CCW, #1 & #4 do not work, so I connected 2 & 3 ESC's w/jumper wires to motors 1 & 4 and they worked fine..? So, I think I missed a set of PINS on the 10 PIN standoff. The I/O PWM OUT on the Pixhawk 4 is a 10 pin connection, with pin 1 being a Red 5v wire.... That threw me off by 1 pin so I started the ESC Signals wires at pin #2 through 6.... I found a diagram on one of the Pixhawk 4 wiring sets where they used the PM02 Power Module and I saw my mistake, they started at Pin # 1 through 4, so I switched it over and it's late now so I'll connect motors 1 & 4 tomorrow after I get my errands done.... I'm pretty sure it'll work now, at Least I hope so.....!....?
James.....

Sent with [ProtonMail](https://protonmail.com) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

hamishwillee, 2020-11-03:
@jimlp12 Howdy. Just to be very clear, now that you have done this none of my previous advice will be useful to you. If you're working on Mission Planner you MUST be using ArduPilot as it does not support PX4. You should ask further questions related to this on the ArduPilot discussion boards.

Good luck!

jimlp12, 2020-11-03:
Hamish, yeah I know that I'm running Arducopter...on Mission Planner, at least it works there...! Not like QGC...!
James....

Sent with [ProtonMail](https://protonmail.com) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐