mavlink/qgc-user-guide #311: Guide directs to private Slack workspace without additional info

aib, 2021-12-10:
At https://docs.qgroundcontrol.com/master/en/Support/Support.html#developer_chat there is a link to a [#QGroundControl Slack channel](https://px4.slack.com/). This seems to be a channel for people with a `@px4.io` e-mail address. As a reader of the guide, I don't know whether (or how) I should get one or somehow get another e-mail address invited to the channel or what.

DonLakeFlyer, 2021-12-10:
You just get a Slack signin and the add then connect to the channel. Open to anyone.

closed 2021-12-10 by DonLakeFlyer

aib, 2021-12-11:
Is there anything on how to do that?

aib, 2021-12-16:
For the record, there is an invitation link at http://slack.px4.io/ currently mentioned in this repository's README and nowhere in the guide. After joining the PX4 Slack workspace, joining the #QGroundControl channel is indeed trivial. (`[+] Add channels` and a few other places in the Slack UI.)