mavlink/qgc-user-guide #312: Guide needs info about how to "get a Slack signin and the add then connect to the channel"

aib, 2021-12-15:
As discussed in #311.

Searching around only leads to more people confused by how to do this:

https://discuss.px4.io/t/slack-invites/22857
https://discuss.px4.io/t/qgc-slack-error/23188

Slack itself is no help and implies it's not possible:

[image]

hamishwillee, 2021-12-15:
Let me just check what you have tried - have you clicked http://slack.px4.io/ ?
That takes me to a place where I can get an invite:
[image]

If that works we could update https://docs.qgroundcontrol.com/master/en/Support/Support.html#developer_chat like this:
[image]

aib, 2021-12-16:
Ah, that was precisely the information I was missing. Thanks, it indeed works.

hamishwillee, 2021-12-16:
Thanks - also for posting the fix!

closed 2021-12-16 by hamishwillee