mavlink/qgc-user-guide #355: "Download and Install" mentions macOS 10.20, but it doesn't exist

mpiccolino-tealblue, 2022-08-10:
https://docs.qgroundcontrol.com/master/en/getting_started/download_and_install.html#macOS

I can't fix it as I don't know what the minimum macOS version is.

hamishwillee, 2022-08-11:
Me either. Previously it was 1.10 Yosemite - @DonLakeFlyer could have meant 1.11 ("next version") or 1.12. @DonLakeFlyer do you know how the minimum version can be worked out? @mrpollo ?

DonLakeFlyer, 2022-08-11:
I'm not exactly sure. But I would guess at best the minimum version of OSX supported is whatever the minimum version support by the current version of Qt we are using is. I don't know if other things like video have different requirements.

hamishwillee, 2022-08-11:
Thanks @DonLakeFlyer @mpiccolino-tealblue - 10.11 based on the build instructions in the dev guide. I added a note about the Qt version being the dependency in the source.

closed 2022-08-11 by hamishwillee