mavlink/qgc-user-guide #393: Video issue specific to Linux on Latitude 5430 Rugged

fniles, 2023-03-13:
We have many Linux laptops running QGroundControl and have tested video on all of them and it works just fine. We are using RTSP streams. We recently purchased two Dell Latitude 5430 Rugged laptops and noticed the video come through on QGroundControl with the wrong Aspect Ratio. That is, the video about 1/4 the width of the actual QGroundControl window. (3/4 of the screen remains black)

Here's some debugging we did:
* The issue doesn't happen on any of our other Dell XPS machines or other linux laptops we have tried.
* We have two of these laptops and reinstalled everything on the second laptop and the same issue DOES occur
* On this SAME laptop it does NOT happen running QGroundControl in Windows
* It also does NOT happen in Linux when running other RTSP clients like gstreamer by hand or VLC.

Obviously, it's something about the support packages, the video hardware on this laptop and some way QGroundControl is configuring gstreamer. However, I'm not sure what to try next.  I will try to build QGroundControl from source code and see if that gives us any clues.

Thanks for any ideas of what to try,
Rick.

hamishwillee, 2023-03-22:
Hi Rick,
No idea. But this area is primarily for bugs in the docs. For support queries, I suggest you ask on discord.

@DonLakeFlyer Can you suggest further debugging steps? Or someone specifically we can ask?

DonLakeFlyer, 2023-03-22:
Sorry I don't know much about video. The ArduSub folks are your best bet since they use it a lot. @patrickelectric might be a good starting place.