mavlink/qgc-user-guide #399: :-1: error: Project ERROR: Unknown module(s) in QT: charts

programmingyug, 2023-11-10:
When, I tried to debug , after importing from github. In Issues " :-1: error: Project ERROR: Unknown module(s) in QT: charts "

Development on :
  Qt Creator 5.12.12
  Visual Studio 64bit 2017
  Windows 11

I am facing this problem since 5 days.
My contact/ WhatsApp : 7079887439

programmingyug, 2023-11-11:
I have already tried on different versions of Qt Creator. i.e. 05.12.12 / 05.15.
But nothing is happening more than that.

hamishwillee, 2023-11-15:
Is this a new error - i.e. have you had this working before? Did you install the charts as per step 2 in https://dev.qgroundcontrol.com/master/en/getting_started/#install-qt ?

I would be asking on the discussion boards.

programmingyug, 2023-11-24:
> Is this a new error - i.e. have you had this working before? Did you install the charts as per step 2 in https://dev.qgroundcontrol.com/master/en/getting_started/#install-qt ?
> 
> I would be asking on the discussion boards.

I am happy to announce that , " I have completed it "  i.e. I have successfully build QGroundControl on Qt Creator.

programmingyug, 2023-11-24:
I have successfully build QGroundControl on Qt Creator.

WhatsApp : 7079887439

closed 2023-11-29 by hamishwillee